dot-bob/Marlin-Duplicator-6 #1: repo sync

Andrei-Pozolotin, 2017-05-21:
Robert:

1) this repo is isolated from https://github.com/MarlinFirmware/Marlin ?

2) do you have plans for upstream/downstream sync with your repo ?

3) if you use private remote, perhaps you could push sync branches ?

Thank you.
Andrei.

Andrei-Pozolotin, 2017-05-21:
asked upstream: https://github.com/MarlinFirmware/Marlin/issues/6818

dot-bob, 2017-05-22:
You can still add Marlin mainline as a remote and update.  I isolated it from the fork as people were getting confused and downloading the wrong branch.  I am working on a release based off 1.1.x hopefully I will get it released soon.  I have it on my Duplicator 6 now just running tests with it.

Andrei-Pozolotin, 2017-05-24:
Robert:

I finally got to the point when I am ready
to try custom marlin firmware

Can you please share:

1) how usable/stable is your patch:
https://github.com/dot-bob/Marlin-Duplicator-6 ?

2) what are the improvements,
compared to the stock firmware ?

3) I tried to upgrade your patch
to current marlin master - no luck so far,
too many changes, the code just wont build
hey, will try again, and if you have any tips - please share

Thank you.

Andrei-Pozolotin, 2017-06-11:
Robert: 
can you please push your current working Wanhao-D6 branch to your marlin fork?
https://github.com/dot-bob/Marlin/branches/active
Thank you.

dot-bob, 2017-06-12:
The current version would not be accepted as a whole to mainline.  I am working on individual submissions to mainline to add basic support for the D6 hardware.

I am holding off on a 1.1.x port until a few bugs are fixed and I get a few features added to mainline.  If you are looking for a port to the latest revision you are welcome to use my personal port I have been using for development.  It is located on the MMU branch of my main marlin fork: https://github.com/dot-bob/Marlin/tree/MMU

One thing to note is I use an auto level sensor on the analog port not the standard end stop connection.  You will need to edit the pins_ULTIMAIN_2.h to use the stock end stop again.

Andrei-Pozolotin, 2017-06-12:
great, thank you

Andrei-Pozolotin, 2017-06-15:
I am curious as to why are you using analog z-probe port, 
are you trying to replicate ultimaker 3 approach? :-)

https://ultimaker.com/en/community/23463-inside-the-ultimaker-3-day-6-active-leveling
Active leveling - Touching the bed
[image]

dot-bob, 2017-06-21:
The IR sensor I am using doesn't like the fixed pull-up on the normal endstop connection.  The analog port doesn't have this pull-up.

closed 2017-06-21 by dot-bob