dot-bob/Marlin-Duplicator-6 #2: Firmware wanhao D6 improvements

MONZlLLA, 2017-06-01:
Hi Dot-Bob, I'm a happy owner of a Wanhao D6.
Up to a week ago I used the Wanhao D6 firmware 3.1.
The printer printed very well without any problems.
Then I came across your Marlin firmware that you did ad hoc for this printer.
Installed and telling you the truth is far superior, both for basic features and for precision and speed of use.
That's why I thank you.
The problem comes now ... I noticed that printing the first layer is not perfect yet or tried with Simplify3D to make multiple settings but I could not fix it.
Now I have to reluctantly reinstall the original firmware and the first Layer and perfect smooth and clean.

I think maybe there is a problem with the alignment of the printing bed that makes some weird jokes.

I ask you that it is possible for you to insert in your firmware the system of the original version that I have now installed.
This is the [link to the original firmware 3.1](https://github.com/garychen99/Duplicator-6-firmware-version-3.1) so you can see that they did these of the wanhao.

Hello and thanks again for your fantastic contribution.

dot-bob, 2017-06-12:
I assume you using the mesh bed leveling.  If so are you saving the mesh after leveling the bed?

dot-bob, 2017-06-21:
There are two types of manual leveling available for the stock D6, standard and mesh.  It is currently defaulting to mesh.  Comment #define MESH_BED_LEVELING to use the standard endstop with no compensation.  Also because marlin supports so many printers it doesn't have a option to locate the head to fixed points for bed level assisting.  You will need to manually position the head to adjust bed screws.

closed 2017-06-21 by dot-bob