dot-bob/Marlin-Duplicator-6 #3: PWM leds

timbit123, 2017-06-28:
Hello,
I saw two led behind the knob, they turn on after I flashed but never came back after a reboot. I don't know if there is a way to get the pins number but we could activate the config"RGB_LED" and TEMP_STAT_LEDS too.

dot-bob, 2017-06-29:
For Marlin RC8 use gcode M150 to set the color of the LEDs.
I just uploaded the 1.1.x branch which is based of the marlin bugfix-1.1.x branch from a few days ago.  This version supports the PRINTER_EVENT_LEDS out of the box

```/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
```
[https://github.com/dot-bob/Marlin-Duplicator-6/tree/1.1.x](https://github.com/dot-bob/Marlin-Duplicator-6/tree/1.1.x)
Let me know if you have any issues.

timbit123, 2017-06-29:
look like its working! I activated this in the config and it light up
#define RGB_LED
//#define RGBW_LED
#if ENABLED(RGB_LED) || ENABLED(RGBW_LED)
  #define RGB_LED_R_PIN 34
  #define RGB_LED_G_PIN 43
  #define RGB_LED_B_PIN 35
  #define RGB_LED_W_PIN -1
#endif

closed 2017-09-01 by timbit123