dot-bob/Marlin-Duplicator-6 #4: Heatbed hit bottom after installing glass bed

yufanyufan, 2017-07-19:
After installing glass bed and re-adjust z home, the heat bed hit the max of z each time changing filament, and may damage the step motor.

I think this is because in the changing filament routine, the machine automatically lower the build plate to 175, with out compensate the thickness of glass.

timbit123, 2017-09-01:
I have the same setup and never had issue, did you unscrew the bolt behind the bed that trigger de switch?

dot-bob, 2018-04-30:
Your ending gcode needs to be modified to prevent it from moving too far.  You could also modify your configuration.h in the firmware to shorten your max z travel and verify software endstops have been enabled.

closed 2018-04-30 by dot-bob