dot-bob/Marlin-Duplicator-6 #5: Compile error Marlin\Marlin.ino:45:110: fatal error: U8glib.h: No such file or directory (#define U8GLIB_SSD1306)

easysofts, 2017-09-29:
My config: Wanhao Duplicator 6 with manual leveling (**no probe**) / original hotend.
**U8glib 1.19.1 is installed on my Windows Arduino software 1.8.4 (lot of verifications)**

I can't test this version because :

**- I can't compile your test branch without error**
- I can't compile earlier Marlin 1.1.5 with D6 Configuration.h file
- But **I can compile default earlier Marlin 1.1.5** (without D6 config)

After compile, It seems to have a bug with the line 1511 in Configuration.h #define U8GLIB_SSD1306.
But If I comment this line no LCD...

**Do you have an idea ?**

Regards,
Mathieu FABRIS
from FB group and personal blog http://www.printer3d.one

###########################

ERROR LOG ON COMPILE 

Arduino : 1.8.4 (Windows 10), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10804 -build-path C:\Users\Mathieu\AppData\Local\Temp\arduino_build_814058 -warnings=all -build-cache C:\Users\Mathieu\AppData\Local\Temp\arduino_cache_798229 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Téléchargements\Marlin-Duplicator-6-1.1.x (1)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10804 -build-path C:\Users\Mathieu\AppData\Local\Temp\arduino_build_814058 -warnings=all -build-cache C:\Users\Mathieu\AppData\Local\Temp\arduino_cache_798229 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Téléchargements\Marlin-Duplicator-6-1.1.x (1)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Mathieu\AppData\Local\Temp\arduino_build_814058\sketch\Marlin.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Mathieu\AppData\Local\Temp\arduino_build_814058\sketch\Marlin.ino.cpp" -o "C:\Users\Mathieu\AppData\Local\Temp\arduino_build_814058\preproc\ctags_target_for_gcc_minus_e.cpp"
In file included from C:\Users\Mathieu\AppData\Local\Temp\arduino_build_814058\sketch\MarlinConfig.h:39:0,

                 from E:\TÃ©lÃ©chargements\Marlin-Duplicator-6-1.1.x (1)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:307: error: #error "BABYSTEP_ZPROBE_OFFSET requires a probe."

     #error "BABYSTEP_ZPROBE_OFFSET requires a probe."

      ^

E:\Marlin-Duplicator-6-1.1.x (1)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino:45:110: fatal error: U8glib.h: No such file or directory

compilation terminated.

exit status 1
#error "BABYSTEP_ZPROBE_OFFSET requires a probe."

dot-bob, 2017-09-30:
Sorry, it was a failed merge that caused the error.  I just fixed the build.  Give it a go and let me know if you have issues.

easysofts, 2017-09-30:
Hello,

I can't compile, same error log with ...\Marlin-Duplicator-6-1.1.x (2)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino:45:110: fatal error: U8glib.h: No such file or directory

It's strange because U8glib 1.19.1 is installed

An idea ?

---

Arduino : 1.8.4 (Windows 10), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10804 -build-path C:\Users\Mathieu\AppData\Local\Temp\arduino_build_794305 -warnings=all -build-cache C:\Users\Mathieu\AppData\Local\Temp\arduino_cache_656270 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Téléchargements\Marlin-Duplicator-6-1.1.x (2)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10804 -build-path C:\Users\Mathieu\AppData\Local\Temp\arduino_build_794305 -warnings=all -build-cache C:\Users\Mathieu\AppData\Local\Temp\arduino_cache_656270 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Téléchargements\Marlin-Duplicator-6-1.1.x (2)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Mathieu\AppData\Local\Temp\arduino_build_794305\sketch\Marlin.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Mathieu\AppData\Local\Temp\arduino_build_794305\sketch\Marlin.ino.cpp" -o "C:\Users\Mathieu\AppData\Local\Temp\arduino_build_794305\preproc\ctags_target_for_gcc_minus_e.cpp"

E:\TÃ©lÃ©chargements\Marlin-Duplicator-6-1.1.x (2)\Marlin-Duplicator-6-1.1.x\Marlin\Marlin.ino:45:110: fatal error: U8glib.h: No such file or directory

compilation terminated.

exit status 1
Erreur de compilation pour la carte Arduino/Genuino Mega or Mega 2560

dot-bob, 2017-09-30:
Manually verify you have the u8glib installed in the correct place.  Your search path/directory layout for the libraries looks non standard.  It is looking for your libraries are located "E:\OneDrive\Documents\Arduino\libraries" and "C:\Program Files (x86)\Arduino\libraries"

My downloaded libraries are located under "Documents\Arduino\libraries" under my user directory.

easysofts, 2017-09-30:
Ok thank you. U8glib was into this 2 folders.
I have deleted "E:\OneDrive\Documents\Arduino\libraries" 
On restart of Arduino Software a popup show that Arduino will use now default library path !

Firmware flashed ok !

closed 2017-12-29 by dot-bob