dot-bob/Marlin-Duplicator-6 #6: Question about the extruder fan

anoldguy, 2018-05-15:
### Description

Looking at Configuration_adv.h I see that the extruder fans are disabled, which makes the extruder fan stay on all the time. I also see in some of the Ultimaker code repos comments that suggest the fan pin for the extruders was removed, implying they switched to an always-on arrangement?

I've got a Monoprice Maker Ultimate, running off the MMU-2.0.x branch (c86943e01).  I'm loathe to take my machine apart to trace these, and I'm hoping there's a clear answer. A "no" just means I'll stop hunting for it.  :smile: 

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

dot-bob, 2018-05-15:
Unfortunately they just hardwired the extruder fan to 24V.  There isn't much free I/O routed from the microprocessor on the mainboard to use with a relay to switch it on and off.  

I am using the analog port for my BL_Touch and I really didn't want to solder directly to the micro so I used a small buck converter, dropped the voltage to 12V, and installed a 40x10 Noctua fan that is almost silent.

anoldguy, 2018-05-15:
1. Thanks for the quick, knowledgeable reply!
2. Thank you for the work in bringing Marlin to the MMU/Wanhao D6!
3. If I have further questions, is a GH issue the best way to ask them?

dot-bob, 2018-05-15:
It is fine to use GH to ask questions.  I am also on facebook as Robert Mendon.

anoldguy, 2018-05-16:
I'm not on facebook (but I am on [keybase.io](https://keybase.io/mr_ndrsn)), so I'll drop questions here.

I've got the all-metal hot-end upgrade from MicroSwiss, and I was looking at attempting to print polycarbonate. Some PC needs to be printed at/near 300C, so I know I'll have to bump MAXTEMP.  Any experience with that?

dot-bob, 2018-05-19:
Yeah just change the HEATER_0_MAXTEMP to something like 310C.   I print polycarbonate on mine all the time.  I printed my cooling fans in it as I got tired of them drooping all the time.

I have my max temp on the hotend set to 400C and the bed set to 150C  but have yet to take it that high.  I have taken the bed up to 140C so far.  I did change the connector on the bed as the stock one was melting at 120C.  If you go much above 300C you will not want to use a silicone sock.

In the next week or so I may try to print some small PEEK parts on it at about 380C.

dot-bob, 2018-05-20:
Check your thermistor and see if it is still secure on the bottom of the bed.  It may be an issue with mainline marlin.  You may want to try manual adjustment (see manual section of [PID Tuning](http://reprap.org/wiki/PID_Tuning)).  

I am using a glass sheet with PEI for my build plate for my D6.  As a matter of fact most of my printers have a large gIass bed or MIC6 aluminum and have a lot of thermal mass so I usually disable PID control for the bed.  If I enable PID control it actually causes more temperature fluctuation as my beds need full power to sustain a constant temperature.