dot-bob/Marlin-Duplicator-6 #7: FIL_RUNOUT_PIN not defined in 2.0 branch

bblonski, 2018-05-20:
Will not compile when FILAMENT_RUNOUT_SENSOR is enabled.  I had to manually copy over the FIL_RUNOUT_PIN definition into pins_Ultimaker_2.h from the RC8 source.

dot-bob, 2018-05-20:
Yeah there really isn't a standard connection for the runout sensor.  Bruce Clark added the it to default to the analog port in RC8,  This change didn't move up mainline as the pins file is used for multiple machines besides the D6.  In fact I am using the analog port for my BLTouch sensor on my machine.  

I am trying to keep the new releases as close to mainline as I can to make it easier to incorporate upstream changes.

You could probably add the FIL_RUNOUT_PIN to your configuration.h instead of the pins file to make it easier to update to future releases.

bblonski, 2018-05-20:
Good point.  It just caught me off guard migrating from RC8 since this repo is geared towards the Duplicator specifically.  Maybe add a commented out line in the Configuration.h so new users don't have to go look up pin numbers?  I also had to disable arc support in Configuration_adv.h to get things to compile.  Is there need to maintain compatibility with mainline Marlin?