dot-bob/Marlin-Duplicator-6 #8: MMU vs D6 Branch Clarification

arctelix, 2019-02-01:
Hi dot-bob.

First, thanks for all your hard work and contributions to the MMU/D6 community!

I just recieved my MMU a few days ago and wanted to clarify a few things.

1) Looks like this is the latest repo for MMU/D6 marlin ports?
https://github.com/dot-bob/Marlin-Duplicator-6/

2) Are there any hardware differenced between MMU and D6?  In other words, is there any reason not to install the latest 1.1.X branch of this repo on an MMU other then the CUSTOM_MACHINE_NAME in the configuration file?

Thanks in advance for any info you can provide!
- A

dot-bob, 2019-02-03:
Initially this branch and the branch for my personal D6 (located in another repository) were used to develop initial support for the duplicator 6 in mainline.  This branch is setup for those who have a stock duplicator 6 and don't want to mess with compiling code or want something that works off the bat.  I have it setup to automatically build when I commit changes.

The branch is a little behind the development branch in mainline as I try to only update when there is something worthwhile and the firmware feels stable.  I usually update it alongside my personal branch so I can test most of the features as my D6 isn't stock anymore.

If you are the type that wants bleeding edge I encourage you to use mainline and the configuration located in the examples included with mainline.

arctelix, 2019-02-03:
Thanks for the reply dot-bob!

As a starting point, can you confirm if there is any hardware differences between the D6 & MMU that would require a configuration change? 

For now i'd like to just get the latest stable branch you have tuned for a stock D6 and make any mods required for the MMU.  Which repo/branch would that be?

Thanks Again!

dot-bob, 2019-02-05:
Hardware wise the Duplicator 6 (D6) and the Monoprice Maker Ultimate (MMU) are the same.  The only difference is the firmware on the MMU has been rebranded.

As far as branches I would go with bugfix-2.0.x.  The branch 1.1.x is end of life and receives few updates.  I will be updating my branch to mainline bugfix-2.0.x in the next week after I have run it a while on my MMU.

cmanske, 2019-02-21:
> Hardware wise the Duplicator 6 (D6) and the Monoprice Maker Ultimate (MMU) are the same. The only difference is the firmware on the MMU has been rebranded.

It sounds like it should be safe to install on my MMU.  Has anyone tested this?

Deneteus, 2019-12-05:
@dot-bob Are you going to update this branch now that Marlin 2.0 stable is out?
The stock Marlin 2.0 Wanhao D6 example config failed due to a sanity check.
We noticed that your build uses TinyLCD and I was able to use the same setup to get my machine updated a PR was made in regards to the issue.